uob-hep-cad/uob-hep-pc065 #2: Add DNF resistor between PGOOD_C2M and enable pin of 1.3V regulator (U11). Add caps etc. to follow data sheet recommendations

DavidCussans, 2023-12-12:
AFC v4 doesn't seem to raise PGOOD_C2M this disables 1.3V regulator U11

Put a DNF ( i.e. normally not present ) zero-ohm resistor between PGOOD_C2M and enable pin of 1.3V regulator (U11). Add caps etc. to follow data sheet recommendations

Looking at datasheet for U11 https://www.monolithicpower.com/en/documentview/productdocument/index/version/2/document_type/Datasheet/lang/en/sku/MP20075/document_id/1225/ it looks like we haven't followed a number of schematic and layout recommendations. Review schematic and follow recommendations.

oxfpete, 2024-02-23:
DNF resistance added.

4.7uF added to VDRV pin of U11 as per datasheet

10uF & 100nF added to DDQ pin of U11 as per datasheet recommendation of Cin being 0.5*Cout

Done.

DavidCussans, 2024-04-04:
Should we also add a 100nF cap to Ref input?

oxfpete, 2024-04-04:
Extra 100nF added to layout without the need to sacrifice the R183 shorting link. Done.

DavidCussans, 2024-04-04:
Looks good on schematic.
Layout looks good.