uob-hep-cad/uob-hep-pc065 #4: Replace Si5345 clock generator with a low-jitter fanout.

DavidCussans, 2023-12-13:
Chaining PLLs increases jitter between endpoints

--> Replace Si5345 clock generator with a low-jitter fanout. One input to clock from AFC. Other input from a 250MHz TCXO (same family but different frequency as used on rest of DTS boards)

oxfpete, 2024-02-23:
Done but have left notes on schematic sheet. Nothing critical just want confirmation is safe to delete bus definitions.

DavidCussans, 2024-02-23:
Which bus definitions?

If they aren’t used, then , yes. ( It would be good not to have signal names that aren’t connected to anything cluttering up the schematic. I would also be good to run the schematic check tool to check for unused/unconnected nets )

Thanks!

DavidCussans, 2024-04-04:
Schematic changes look good.

closed 2024-04-04 by DavidCussans