uob-hep-cad/uob-hep-pc065 #10: Remove jumper J39 and connect P2V5 direct to VAUX ( via fuse - see issue #9)

DavidCussans, 2024-02-07:
Remove jumper J39 and connect P2V5 direct to VAUX ( via fuse - https://github.com/uob-hep-cad/uob-hep-pc065/issues/9 )
Attach test point (TP10) to P2V5

oxfpete, 2024-02-23:
Added RS692-5003 to cnpassive.ptf but doesn't appear under any "fuse" library. Instead is available under cnpower as "BOURNS_SF_0603S300_2". Packages fine and has the related part number for a BOM but not sure how this library was originally set up. Have only ever seen ptf files within each library subfolder, maybe I have missed something on the setup.

Done - but not convinced I have added the fuse properly to system library ptf.

DavidCussans, 2024-02-23:
CERN and (from what I remember RAL) use multi-part part tables for some items (resistors, inductors, capacitors, etc. ) and a per-part part table for other items (things like ASICs etc. )

My understanding is that there is no harm in having a part like BOURNS_SF_0603S300_2 with it’s own part table.

Could you move the new fuse component BOURNS_SF_0603S300_2 from cnpassive to cds_discrete ? The cnxxxxx libraries are copies of CERN Cadence libraries and I try not to make any changes here . The cds_xxxx libraries are home-brew so I put components I create here. It is fine to have a single item part-table under BOURNS_SF_0603S300_2

[​loop icon] RE [uob-hep-caduob-hep-pc065] Remove jumper J39 and connect P2V5 direct to VAUX ( via fuse - see issue 9) (Issue 10).loop<https://uob-my.sharepoint.com/:fl:/g/personal/phdgc_bristol_ac_uk/EcKcPLQbCz9HtGLS8m59YEYB0c1AIczKHVs7nayqeSeAHg?nav=cz0lMkZwZXJzb25hbCUyRnBoZGdjX2JyaXN0b2xfYWNfdWsmZD1iIXBoWFpyMlhsMVVhWC1vUW8zZVZQR1F4MjdjdGxQTE5PckpKZi1OWXNqZUp3ZlhsbUJyblNSS083cmFOQTBxcFgmZj0wMTJLRDJFRldDVFE2TElHWUxINUQzSVlXUzZKWEgyWUNHJmM9JTJGMTUzMTJhNzgtZTM2ZS00YzQ1LWI4NWQtNjRkYjA0ZWE4NWZjJmZsdWlkPTEmcD0lNDBtcyUyRm9mZmljZS1mbHVpZC1jb250YWluZXI%3D>

DavidCussans, 2024-04-04:
Schematic change looks good.
I haven't tried a physical export to make sure that the library is still good.

DavidCussans, 2024-04-24:
Looks good on PCB as well as schematic. Closing issue

closed 2024-04-24 by DavidCussans