uob-hep-cad/uob-hep-pc065 #36: Route signals from SFPs to clock capable pins.

DavidCussans, 2024-05-08:
The signal from the SFPs , RD_FPGA_{P,N}<0..7> should all be connected to clock capable pins ( we have changed the firmware with respect to the first FIB prototype and now need to connect the incoming signal to a BUFR ). Connect up the following pair, choose the order that gives the best routing.

clock region X0Y4, bank 16
FMC_HA_00 (CC) -> H29/J29 (MRCC_16)
FMC_HA_17 (CC) -> H28/J28 (MRCC_16)

clock regionX1Y2, bank 34
FMC_HB_00 (CC) -> W5/Y5 (MRCC_34)
FMC_HB_17 (CC) -> AA4/AA5 (MRCC_34)

X1Y3, bank 35
FMC_LA_17 (CC) -> T4/T5 (SRCC_35)
FMC_LA_18 (CC) -> P3/P4 (SRCC_35)

X1Y4: bank 36
FMC_LA_00 (CC) -> K6/K7 (SRCC_36)
FMC_LA_01 (CC) -> G4/G5 (MRCC_36)

oxfpete, 2024-05-16:
Done but will need you guys to check over to make sure FPGA happy with pin assignment

DavidCussans, 2024-05-16:
Brilliant!

I'm in CERN at the moment but will get onto this tomorrow.

Dr David Cussans                    University of Bristol,
office: +44-(117)-95 46879   H.H. Wills Physics Laboratory,
     lab: +44-(117)-33 17199   Tyndall Avenue,
     fax: +44-(117)-925 5624   Bristol BS8 1TL, UK

DavidCussans, 2024-06-03:
Schematic checked by Diana Antic. Believed to be OK.

closed 2024-06-03 by DavidCussans